KULeuven-MICAS/zigzag #2: Missing temporal ordering for temporal ordering conversion on resnet18 example

cstutor888, 2022-06-22:
I tried to run the example, python main.py --workload inputs.examples.workloads.resnet18 --accelerator inputs.examples.hardware.eyeriss_like_simple, and got the following message:

2022-06-22 11:39:29,502 - classes.stages.MainInputParserStages.parse_workload_from_path +44 - INFO - Created workload graph with 28 nodes and 35 edges.
2022-06-22 11:39:29,503 - classes.stages.MainInputParserStages.parse_accelerator_from_path +16 - INFO - Parsed accelerator with cores [1].
2022-06-22 11:39:29,503 - classes.stages.SpatialMappingConversionStage.convert_user_spatial_mapping +98 - INFO - User-provided spatial mapping converted to: {'D1': ('K', 12.8), 'D2': ('C', 3.0)}
2022-06-22 11:39:29,504 - classes.stages.TemporalOrderingConversionStage.check_layer +41 - CRITICAL - Layer LayerNode_0 has no user-defined spatial mapping.
Traceback (most recent call last):
  File "/home/nesong/Projects/zigzag-v2/main.py", line 29, in <module>
    mainstage.run()
  File "/home/nesong/Projects/zigzag-v2/classes/stages/Stage.py", line 53, in run
    for cme, extra_info in self.list_of_callables[0](self.list_of_callables[1:], **self.kwargs).run():
  File "/home/nesong/Projects/zigzag-v2/classes/stages/MainInputParserStages.py", line 75, in run
    for cme, extra_info in sub_stage.run():
  File "/home/nesong/Projects/zigzag-v2/classes/stages/SaveStage.py", line 34, in run
    for id, (cme, extra_info) in enumerate(substage.run()):
  File "/home/nesong/Projects/zigzag-v2/classes/stages/WorkloadStage.py", line 25, in run
    for cme, extra_info in sub_stage.run():
  File "/home/nesong/Projects/zigzag-v2/classes/stages/SpatialMappingConversionStage.py", line 59, in run
    sub_stage = self.list_of_callables[0](self.list_of_callables[1:], **kwargs)
  File "/home/nesong/Projects/zigzag-v2/classes/stages/TemporalOrderingConversionStage.py", line 22, in __init__
    self.check_layer(layer)
  File "/home/nesong/Projects/zigzag-v2/classes/stages/TemporalOrderingConversionStage.py", line 42, in check_layer
    raise ValueError("Missing temporal ordering for temporal ordering conversion")
ValueError: Missing temporal ordering for temporal ordering conversion

Could you please check it again?

asyms, 2022-06-22:
Hi,

The reason you're getting this error, is because you have the `TemporalOrderingConversionStage` in your list of stages to be ran, but you haven't provided the Temporal loop order for all the resnet18 layers. The provided resnet18 workload doesn't include these temporal loop orders. 

The easiest thing to do is **comment out** this `TemporalOrderingConversionStage` (L19 in main.py) and **uncomment** the `LomaStage` (L20 in main.py). That way, the `LomaStage` will automatically generate temporal mappings and do a cost model evaluation for each temporal mapping generated.

If you don't want the tool to save all of these cost model evaluations, you can insert a `MinimalEnergyStage` or `MinimalLatencyStage` right above the `LomaStage`. In this way, only the best cost model evaluation is saved and passed up the chain to the `CompleteSaveStage` (which saves the json).

I have also made the error message more detailed. I Hope this clears up the confusion?

KR,
Arne

cstutor888, 2022-06-22:
Thanks, it works.