KULeuven-MICAS/zigzag #12: Why is the number of dimension for the MatMul ONNX parser fixed to 2

MaxenceBouvier, 2023-07-05:
Hello,

I would like to understand why the MatMul ONNX parser is asserting that the length of the input and output dimension of the operand is set to 2:
https://github.com/KULeuven-MICAS/zigzag/blob/b3e0c04782516441e8250fbb0cd89573b8f10da6/zigzag/classes/io/onnx/matmul.py#L65

As far as I understand, a MatMul would often have two operands of 3 dimensions, of shapes (batch_size, x, y).
The MatMulParser expects the MatMul operands to find the batch_size at dimension 0 anyway.
https://github.com/KULeuven-MICAS/zigzag/blob/b3e0c04782516441e8250fbb0cd89573b8f10da6/zigzag/classes/io/onnx/matmul.py#L66

I would suggest replacing the following.
In the `assert` above, change ` == 2` to ` in [2,3]`.

LY-Mei, 2023-07-05:
We assume that all operands of MatMal are 2 dimensions: Input, weight, and output are all 2D matrices, and in total MatMul is a 3-dimensional operation.

for b in 0 to B:
  for k in 0 to K:
    for c in 0 to C:
      O[b][k]+=B[k][c]*A[b][c]

Does this solve your doubt?

MaxenceBouvier, 2023-07-05:
Thanks for fast answer.
Yes I understand the assumption made.

So, it means that I should modify my ONNX workload to represent this Batched 2D*2D MatMul operation (batch_size=N) as N different 2D*2D MatMul operations?

asyms, 2023-07-05:
Hi,

Ideally it's defined as one layer so the mapping optimization engines can take advantage of all the reuse across batches. Feel free to open a PR that is able to handle both 2D or 3D input tensors, or customize the parser to your liking.

MaxenceBouvier, 2023-07-06:
Ok great thanks a lot.

To try to put me on the right track,
Could please you tell me if ZigZag would manage correctly this parser, or would it be an issue?
The underlying doubt is that I don't know how "flexible" ZigZag can be regarding the equation, the number of dimensions, etc.

Case:
Weight matrix W is of shape (WY, WX)
Input matrix I is of shape (B, IY, IX) - where B is the batch size.

````python
# Equation
d = {}
d["equation"] = "O[b][iy][wx]+=W[wy][wx]*I[b][iy][ix]"

# Get dimension sizes from input parameters
B = B
IY = IY
IX = IX
WY = WY
WX = WX

d["loop_dim_size"] = {"B":B, "IY":IY, "IX":IX, "WY":WY, "WX":WX}
d["dimension_relations"] = []
d["operand_precision"] = {"O": 16, "O_final": 8, "W": 8, "I": 8}
d["operand_source"] = {"W": [], "I": []}
d["constant_operands"] = ["W"]
d["memory_operand_links"] = {"O": "O", "W": "I2", "I": "I1"}
```

Thanks a lot.

MaxenceBouvier, 2023-07-07:
Ok, it seems to be working fine.

I needed to change:
https://github.com/KULeuven-MICAS/zigzag/blob/b68a9bb07bf89743cc6265878a26cef3f078d629/zigzag/classes/io/onnx/matmul.py#L45

To:
```python
d["spatial_mapping"] = node_mapping.get("spatial_mapping", None)
```

closed 2023-07-07 by MaxenceBouvier

asyms, 2023-07-07:
Great! Glad to hear it's working. The parser you defined looks good to me.

zzczzc20, 2023-08-15:
Hi everyone, I need a special 3D matmul, and all variables have batch dimension (which is described in onnx/docs/Operators.md). For each batch, it is just a common 2D matmul.
Can zigzag support such an operator? Is the parser below ok?

``` python
d = {}
d["equation"] = "O[b][iy][wx]+=W[b][wy][wx]*I[b][iy][ix]"

# Get dimension sizes from input parameters
B = B
IY = IY
IX = IX
WY = WY
WX = WX

d["loop_dim_size"] = {"B":B, "IY":IY, "IX":IX, "WY":WY, "WX":WX}
d["dimension_relations"] = []
d["operand_precision"] = {"O": 16, "O_final": 8, "W": 8, "I": 8}
d["operand_source"] = {"W": [], "I": []}
d["constant_operands"] = ["W"]
d["memory_operand_links"] = {"O": "O", "W": "I2", "I": "I1"}
```

MaxenceBouvier, 2023-08-15:
It's going to work but there is an issue, one of the dimension of the matrix multiply operation is "collapsing".
So, it should get the same coordinate notation.
ix == wy --> s
d["equation"] = "O[b][iy][wx]+=W[b][s][wx]*I[b][iy][s]"