KULeuven-MICAS/zigzag #14: Layer Assigned to CME is Passed by Reference

MaxenceBouvier, 2023-07-12:
Hi,

In the spatial mapping generator, the layer assigned to the CME is passed by reference.
Which means that all returned CMEs will point to the same layer at the end of the full ZigZag stages loop.

At this line:
https://github.com/KULeuven-MICAS/zigzag/blob/de6d4bbecec8cbb1f108ad2ed379ab7cfef01d0a/zigzag/classes/stages/SpatialMappingGeneratorStage.py#L89

So the final selected CME that is saved during a save stage may not point to the actual layer it was generated from.

I would suggest to assign a copy of the layer to make sure that each CME yielded by the SpatialMappingGenerator have the correct layer as attribute.
```python
import copy
[...]

spatial_mapping_conversion_stage = SpatialMappingConversionStage(
                [...]
                layer=copy.copy(self.layer),
                **self.kwargs,
            )
```

MaxenceBouvier, 2023-07-12:
Update:
See pull request [here](https://github.com/KULeuven-MICAS/zigzag/pull/15).

asyms, 2023-09-19:
PR has been merged.

closed 2023-09-19 by asyms