KULeuven-MICAS/zigzag #22: How can I use the ARCHITECTURE GENERATOR mentioned in the paper of zigzag

morainer, 2023-09-15:
I can not find any code or document that describe the details of the ARCHITECTURE GENERATOR. Could you offer an example for me?
PS: The paper of zigzag is https://ieeexplore.ieee.org/document/9360462#:~:text=This%20article%20introduces%20ZigZag%2C%20a%20rapid%20DSE%20framework,uneven%20mapping%20opportunities%20and%20smart%20mapping%20search%20strategies.

morainer, 2023-09-18:
[image]

BTW, according to the calculating method mentioned in the paper, I think that the MAC operation inside of each PE should be 320 but not 1600, as the FYu 5 LOOP is a spatial reuse factor and it should not be considered. Please let me know where I got it wrong.

asyms, 2023-09-19:
Hi,

The architecture generator as described in the original paper isn't directly available as a Stage but can be created to your liking by modifying the OperationalArray and MemoryHierarchy into a new Core and Accelerator. You can read more [here](https://kuleuven-micas.github.io/zigzag/hardware.html) and an example of modifying the architecture can be found [here](https://github.com/KULeuven-MICAS/zigzag/blob/master/zigzag/classes/stages/PEArrayScalingStage.py), where the PE array is scaled with a given factor.

Regarding the MAC count inside each PE, I believe you are correct. @LY-Mei can you confirm?

morainer, 2023-09-19:
Very appreciate to receive your answers. I will wait for @LY-Mei ‘s confirmation. And, would you like to offer me an example to use the PEArrayScalingStage?

asyms, 2023-09-19:
You can use it through this API call [here](https://github.com/KULeuven-MICAS/zigzag/blob/5dc9e1d6b450998fcea6fe824ac12e664343680a/zigzag/api.py#L80).

LY-Mei, 2023-09-19:
Yes, you are right. The MAC operation inside of each PE should be 320. Thanks for catching it!

closed 2023-09-20 by asyms

morainer, 2023-09-21:
[image]

Hello,

According to the ZigZag Paper, the memory access energy is calculated by multiplying the memory access count with the corresponding memory per-data-access energy we gived previously. In the Code, I found that **the memory access count** is represent by mapping.combined_mapping.Mapping.unit_mem_data_movement.**data_elem_move_count**. When calculating the memory access energy, the Code used other variables (like **data_trans_amount_per_period**, **data_precision**, **data_trans_period_count**, **min_bw** and **CostModelEvaluation.mapping.spatial_mapping.unit_count**), the major element(**data_elem_move_count**) is not being used however. So my question is : Why does **data_elem_move_count** not need to participate in the **memory access energy**'s calculation?

Thanks for your kind attention and look forward your prompt reply.