KULeuven-MICAS/zigzag #23: Question About Calculating the Memory Access Energy

morainer, 2023-09-21:
[image]

Hello,

According to the ZigZag Paper, the memory access energy is calculated by multiplying the memory access count with the corresponding memory per-data-access energy we gived previously. In the Code, I found that **the memory access count** is represent by mapping.combined_mapping.Mapping.unit_mem_data_movement.**data_elem_move_count**. When calculating the memory access energy, the Code used other variables (like **data_trans_amount_per_period**, **data_precision**, **data_trans_period_count**, **min_bw** and **CostModelEvaluation.mapping.spatial_mapping.unit_count**), the major element(**data_elem_move_count**) is not being used however. So my question is : Why does **data_elem_move_count** not need to participate in the **memory access energy**'s calculation?

Thanks for your kind attention and look forward your prompt reply.

morainer, 2023-09-21:
BTW, if **data_elem_move_count** does not need to participate in **the memory access energy**'s calculation, what is the purpose of calculating **data_elem_move_count**?

LY-Mei, 2023-09-21:
Hi morainer, this is a good question! A short answer to your question is that we did what you thought in the beginning and realized that it can cause inaccuracy in the energy estimation in some cases, and thus we changed it to the current version.

In more detail: In the early version of ZigZag implementation, the energy consumption was indeed calculated based on the data element movement count (multiply it with the average per-data-element-access energy). The average per-data-element-access energy is calculated by dividing memory word access energy by the number of data elements a word can hold. For example, assume 1) a memory has a wordlength (or bandwidth) of 128 bits, 2) read such a 128-bit word from the memory consumes 100pJ, and 3) a data element's precision is 8 bits. The average per-data-element-access energy is, in this case, 100/(128/8)=6.25pJ. If the memory needs to read out 1536 data elements (determined by the mapping), then the read energy consumption of this memory is 1536*6.25=9600pJ.

However, in this way, it can come with certain inaccuracies due to the ignorance of the possible mismatch between data movement behavior and memory bandwidth. For example, if these 1536 data elements are read out from the memory at a rate of 12 data elements per read (this case can happen due to non-ideal spatial mapping for example), then the memory bandwidth will be underutilized (ideally the memory can read out 128/8=16 data elements per time, but now only 12 data elements are useful data). In this situation, following the original way (always assuming an ideal memory bandwidth utilization) to calculate memory read energy is underestimated.

 So in the current ZigZag version, we have improved it by taking more detailed data movement information into account, such as the ones you have listed (data_trans_amount_per_period, data_precision, data_trans_period_count, min_bw and CostModelEvaluation.mapping.spatial_mapping.unit_count). Calculating in this new way takes into account the data transfer rate and other more practical factors so as to get a more accurate energy estimation. The original example in this way will get the read energy: 1536/12*100=12800pJ instead of 9600pJ.

The reason we still keep the data_elem_move_count is that it is still a piece of very informative information for framework debugging and mapping optimality analysis.

morainer, 2023-09-22:
Thank you for your answer. I have a rough understanding, but I still have the following questions:
1) Is the data_elem_ move_ count completely unnecessary for energy calculation now? 
2) Is there any formula or description of the energy calculation method currently used that can provide reference?

LY-Mei, 2023-09-22:
Yes, data_elem_move_count is not directly used for energy calculation now. We are using more detailed data transfer information like how often and how much data is transferred on which memory port and so on to get more accurate energy estimation.
The new version of formula is only in the code itself currently.

morainer, 2023-09-22:
Okay, thank you for your answer. I will look up the code to find the real factors that affect energy calculation.

closed 2023-09-22 by morainer