KULeuven-MICAS/zigzag #27: How can I validated a mapping found by Timeloop in ZigZag?

morainer, 2023-10-19:
How can I validated a mapping found by Timeloop in ZigZag? I can only set the spatial mapping and temporal mapping respectively. But I find that I can not allocate the memory level at the exact loop of the mapping as I wish. So I can not get the result that Zigzag can obtain energy that are approximate to the Timeloop.

asyms, 2023-10-25:
Hi,

I don't understand your question completely. Could you provide a [MWE](https://stackoverflow.com/help/minimal-reproducible-example)? 

The ZigZag spatial and temporal mapping representations work on a memory-level basis, so if you hardcode the loops in each level, that should achieve what you're after?

Kind regards,
Arne

morainer, 2023-10-25:
Hi,

Sorry, maybe my expression is wrong.

When we obtain a mapping in TimeLoop, I want to transform this timeloop mapping into Zigzag way, so that I can run the same mapping both in TimeLoop and Zigzag and then get the results of them. However, when I divide the timeloop mapping into spatial and temporal mappings and input them into zigzag and run it in debug mode, I observe that the loops on each memory level do not align with the timeloop mapping. So the energy results of timeloop and zigzag are quite different.

Therefore, I cannot reproduce the result that Zigzag can obtain energy that are approximate to the Timeloop, which mentioned in the verification paper of zigzag. As shown in the figure below.

[image]

I wonder that if there are other ways to solve such problem？

asyms, 2023-10-26:
I understand what you're trying to achieve, but can you elaborate on "I observe the loops on each memory level do not align with the timeloop mapping"? A MWE will help me pinpoint the difference.

morainer, 2023-10-31:
[tlzz.zip](https://github.com/KULeuven-MICAS/zigzag/files/13211687/tlzz.zip)
Okay, here is an example that I try converting a Timeloop mapping into a Zigzag mapping, but I'm not sure if the conversion is correct. If it is wrong, could you please help me point out the error. Thank you very much.

JiacongSun, 2024-03-14:
Hi morainer,

I've noticed several errors in your configurations:

- The processing element (PE) counts differ between the two frameworks: **168** PEs for Timeloop and **96** PEs for Zigzag.
- The PE array dimension definitions are different: **2D (14x12)** for timeloop, **4D (3x4x4x2)** for zigzag.
- You've used an incorrect bandwidth value for memory instances in Zigzag. Timeloop's bandwidth unit is _**per word**_, whereas Zigzag's is _**per bit**_. To rectify this, you should multiply the value used for Timeloop by the word bit length (32 in your case) for Zigzag.
- The temporal mapping defined for both frameworks are different. It seems you may have misinterpreted some output dimensions from Timeloop. For instance, in Timeloop, the dimension Q represents the input/output kernel size. Therefore, `for Q in [0:4)` in Timeloop's mapping output, it should be interpreted as `('OX', 4) `instead of `('FY', 4)`.

Please review these discrepancies and adjust your settings accordingly. Let me know if you need further assistance.

Best regards,
Jiacong

asyms, 2024-03-25:
Closing due to inactivity

closed 2024-03-25 by asyms