KULeuven-MICAS/zigzag #30: Enquiry about the measurement units of analysis report

keneoneth, 2023-10-26:
Hello there, Zigzag is really an interesting work. I have just started doing some testing with Zigzag and I have a question. What are the measurement units of the energy and latency in the output json report? Are they in Joule and Cycle respectively? Thanks.

asyms, 2023-10-26:
Hi,

Thanks for your interest!

The latency is indeed reported in number of clock cycles (based on the bandwidth being in clock cycles).
The energy technically is unitless. It depends on the energy units of the architecture definition (memory read and write costs and operational cost). Typically those are defined in pJ, but this will heavily depend on your technology/implementation.

keneoneth, 2023-10-30:
@asyms Thanks for your reply! I am also interested in knowing what reference materials are used for getting the information (e.g. parameters like r_cost, w_cost, r_bw, w_bw, latency, etc., as well as operands served in each memory instance and the linkage of read/write ports between them) presented in these code https://github.com/KULeuven-MICAS/zigzag/blob/master/zigzag/inputs/examples/hardware/TPU_like.py#L15-L139? Do they reflect a particular version of TPU in the market? Thanks for the help again!

asyms, 2023-11-07:
The parameters of the TPU_like architecture are obtained as much as possible from reference 2 [here](https://kuleuven-micas.github.io/zigzag/hardware.html). The read and write costs are typically not published separately, so this is determined to the best of our ability using internal and external memory resources.
Also keep in mind that while the costs might not be 100% correct, the relative differences between mappings can still be compared to obtain the best one.

closed 2023-11-07 by asyms