KULeuven-MICAS/zigzag #45: Why PickleSaveStage can't be called in zigzag-imc section even if these parts are intergrated in zigzag?

tony-liu1996, 2024-03-19:
I want to use the interface energy, latency, cme = get_hardware_performance_zigzag(workload,
                                                       accelerator,
                                                       mapping,
                                                       opt='latency',
                                                       dump_filename_pattern="outputs/{datetime}.json",
                                                       pickle_filename="outputs/list_of_cmes.pickle")
However, I can only get the dump_file in json format when I run the onnx workloads on aimc or dimc hardware, so how can I call the get_hardware_performance_zigzag interface only in dump_file in json format?
thanks!

JiacongSun, 2024-03-19:
Hi,

I didn't fully understand the meaning of your description. However, if you run the API without making any changes, the pickle file will be saved at `outputs/list_of_cmes.pickle`. Please note that the information for all layers is saved within this pickle file, rather than saving each layer's information as a separate pickle file (as is done with JSON files).

It would be helpful if you could provide some context about what you are doing and what changes you have made.

tony-liu1996, 2024-03-20:
> Hi,
>
> I didn't fully understand the meaning of your description. However, if you run the API without making any changes, the pickle file will be saved at `outputs/list_of_cmes.pickle`. Please note [...]

Thanks for replying me so quickly! 
Hello, JiacongSun. I want to use aimc and dimc simulations on example models and visualize the performance png generated by your end-to-end api, but I cannot run main-onnx.py successfully, the error is either "list out of range" or "cannot unpack non-iterable LayerNode object" on this code "all cmes = [cme for (cme, extra) in extra_info]" in SaveStage.py, hence I cannot get the pickle file and json file both.

tony-liu1996, 2024-03-20:
> Hi,
>
> I didn't fully understand the meaning of your description. However, if you run the API without making any changes, the pickle file will be saved at `outputs/list_of_cmes.pickle`. Please note [...]

What's more, I met with an exception"zigzag.classes.opt.temporal.loma.engine.NoValidLoopOrderingFoundException: No valid loop ordering was found for layer LayerNode_4. Please make sure the spatial mapping is compatible with the architecture." when I run the main-onnx.py, I comment out the searchunusedmemorystage and removeunusedmomorystage on the main code and get the  layer LayerNode_4.json, but when I use the api "get hardware performance zigzag()",the same exception happened again, and I don't know how to solve this exception, thanks!

JiacongSun, 2024-05-26:
Close due to no response.