KULeuven-MICAS/zigzag #47: Permit Loop Dimension Iterators to have digit suffixes

EmilySillars, 2024-04-08:
**We would really like to express loop dimension iterators with names like d0, d1, d2, but this is not currently possible.**
For ex, given 
- hardware description [here](https://github.com/xdslproject/linalg-to-stream/blob/main/inputs/hardware/snax_gemm.py)
- mapping as follows:
```
mapping = {
    "default": {  # default
        "spatial_mapping": {"PE_ARRAY_D1": ("D0", 8), "PE_ARRAY_D2": ("D1", 8), "PE_ARRAY_D3": ("D2", 8)},
        "temporal_ordering": [ ],
        "core_allocation": 1,
        "memory_operand_links": {
            "O": "O",
            "W": "I2",
            "I": "I1",
        },
    },
}
```
- and workload as follows:
```
workload = {
    0: {
        "operator_type": "default",
        "equation": "O[d0] += I[it] * W[d1]",
        "dimension_relations": ["it=d0+d1"],
        "loop_dim_size": {"D0": 16, "D1": 3},
        "operand_precision": {"O": 32, "O_final": 32, "W": 32, "I": 32},
        "operand_source": {"W": [], "I": []},
        "constant_operands": ["I", "W"],
        "padding": {},
    }
}
```

**The following ZigZag api call fails**
```
zigzag.api.get_hardware_performance_zigzag(workload, accelerator, mapping, "latency", "outputs/my-output.json","outputs/list_of_cmes.pickle")
```
**with error:**
```
    len(scaling_factors) == 2
AssertionError: Please remove any constants in the equation relation iJ=d0+d1.
```

**Suggested Solution:** 
- change [this section of code](https://github.com/KULeuven-MICAS/zigzag/blob/882a7f06554909cbb57029f379406869bfd5a30a/zigzag/classes/workload/layer_node.py#L284C1-L303C54) so that loop dimension iterators with digit suffixes are permitted.

https://github.com/KULeuven-MICAS/zigzag/blob/882a7f06554909cbb57029f379406869bfd5a30a/zigzag/classes/workload/layer_node.py#L284C1-L303C54

JiacongSun, 2024-04-09:
Hi,

Please avoid using layer dimension names containing digits in your iterator. Currently, only dimension names consisting solely of English characters are supported. Otherwise, the layer dimension name may not be identified correctly.

Change your workload to the following will work:
```
workload = {
    0: {
        "operator_type": "default",
        "equation": "O[do] += I[it] * W[dw]",
        "dimension_relations": ["it=do+dw"],
        "loop_dim_size": {"DO": 16, "DW": 3},
        "operand_precision": {"O": 32, "O_final": 32, "W": 32, "I": 32},
        "operand_source": {"W": [], "I": []},
        "constant_operands": ["I", "W"],
        "padding": {},
    }
}
```

If you do require using digits in the dimension name, you can revise the regression formula at: https://github.com/KULeuven-MICAS/zigzag/blob/882a7f06554909cbb57029f379406869bfd5a30a/zigzag/classes/workload/layer_node.py#L273

asyms, 2024-04-15:
Hi @EmilySillars,

I have updated the code to support this type of dimension definition.

Please let me know if you have any other issues.

Kind regards,
Arne

closed 2024-04-15 by asyms