KULeuven-MICAS/zigzag #52: Reducing memory size leads to decrease in load cycles

LordScarface, 2024-05-10:
Hello and thank you for creating this great open source tool!

I am currently trying to model my accelerator in ZigZag for DSE, but I encountered an issue when running ZigZag with different memory sizes. To check if my implementation is the problem, I recreated the issue with the Eyeriss implementation from the repository. 

<details>
<summary>This is my test code for Eyeriss</summary>

```python
from zigzag import api
import zigzag

import os
from zigzag.classes.hardware.architecture.memory_hierarchy import MemoryHierarchy
from zigzag.classes.hardware.architecture.memory_level import MemoryLevel
from zigzag.classes.hardware.architecture.operational_unit import Multiplier
from zigzag.classes.hardware.architecture.operational_array import MultiplierArray
from zigzag.classes.hardware.architecture.memory_instance import MemoryInstance
from zigzag.classes.hardware.architecture.accelerator import Accelerator
from zigzag.classes.hardware.architecture.core import Core

from zigzag.classes.stages import *
from zigzag.classes.cost_model.cost_model import CostModelEvaluation
from typing import Type
import re
from onnx import ModelProto

import pickle
from zigzag.visualization.results.plot_cme import bar_plot_cost_model_evaluations_breakdown

import numpy as np
import matplotlib.pyplot as plt
import itertools

from dataclasses import dataclass

@dataclass
class Eyeriss_config:
    DIM: int = 4
    
    mem0_kb: int = 8
    mem1_kb: int = 64
    mem2_kb: int = 1024

    def __init__(self, **kwargs): 
        self.__dict__.update(kwargs)
    
    def get_mem0_bits(self) -> int:
        return self.mem0_kb * 8000

    def get_mem1_bits(self) -> int:
        return self.mem1_kb * 8000

    def get_mem2_bits(self) -> int:
        return self.mem2_kb * 8000

class Eyeriss():
    def __init__(self, cfg):
        self.cfg = cfg
        cores = self.cores_eyeriss()
        acc_name = "Eyeriss"
        self.eyeriss = Accelerator(acc_name, cores)
        
    def get(self):
        return self.eyeriss
        
    def memory_herarchy_eyeriss(self, multiplier_array, visualize=False):
        """Memory hierarchy variables"""
        """ size=#bit, bw=(read bw, write bw), cost=(read word energy, write work energy) """

        rf1 = MemoryInstance(
[... 269 lines cut ...]
print(cme_for_all_layers[-1].data_loading_cycle)

```
</details>

Now when I run this with the default memory sizes: `eyeriss = Eyeriss(Eyeriss_config(DIM=4,mem0_kb=8, mem1_kb=64,mem2_kb=1024))` I get:
latency_total2 = 260519
data_loading_cycle = 5020

But when I run it with: `eyeriss = Eyeriss(Eyeriss_config(DIM=4,mem0_kb=1, mem1_kb=2, mem2_kb=4))` I get:
latency_total2 = 250439
data_loading_cycle = 220

So decreasing the memory size leads to way fewer cycles for data loading, which seems very counter intuitive to me. I am using the latest version of ZigZag from the master branch. 
Maybe I messed up somewhere in my configuration?

However when looking at the evaluation breakdown for both configs, the energy usage goes way up (as I would expect):
`eyeriss = Eyeriss(Eyeriss_config(DIM=4,mem0_kb=8, mem1_kb=64,mem2_kb=1024))`:
[image]
`eyeriss = Eyeriss(Eyeriss_config(DIM=4,mem0_kb=1, mem1_kb=2, mem2_kb=4))`:
[image]

I would love some input on this,
Best Regards,
Lukas

JiacongSun, 2024-05-23:
Hi Lukas,

The `data_loading_cycle` is derived from the temporal mapping in the output. You can check the difference between the generated temporal mapping in the two cases.

One possible reason for your case is you are using EDP instead of latency as the optimization target. To analyze in more depth, you can try to rerun zigzag by fixing the temporal mapping of the second case, to see if the EDP will be higher than the current result.

Best regards,
Jiacong

LordScarface, 2024-05-26:
Hey Jiacong,

thank you very much for the reply! I ran it again with "latency" as the optimisation metric, but the results are exactly the same.

For the mapping I get the following with `eyeriss = Eyeriss(Eyeriss_config(DIM=4,mem0_kb=1, mem1_kb=2, mem2_kb=4))`:
```
********* Temporal Mapping - CostModelEvaluation(layer=LayerNode_0, core=1) *********
 O (O): [[], [('B', 5), ('K', 5), ('C', 5)], [], [('B', 20), ('K', 5), ('C', 5)]]    
 W (I2): [[('B', 5), ('K', 5)], [('C', 5), ('B', 20)], [('K', 5), ('C', 5)]]         
 I (I1): [[('B', 5), ('K', 5)], [('C', 5)], [('B', 20), ('K', 5), ('C', 5)]]         
                                                                                     
-------------------------------------------------------------------------------------
 Temporal Loops                  O                  W                  I             
-------------------------------------------------------------------------------------
 for C in [0:5)                  dram               dram               dram          
-------------------------------------------------------------------------------------
  for K in [0:5)                 dram               dram               dram          
-------------------------------------------------------------------------------------
   for B in [0:20)               dram               sram_64KB          dram          
-------------------------------------------------------------------------------------
    for C in [0:5)               sram_8KB           sram_64KB          sram_1M       
-------------------------------------------------------------------------------------
     for K in [0:5)              sram_8KB           rf_64B             rf_64B        
-------------------------------------------------------------------------------------
      for B in [0:5)             sram_8KB           rf_64B             rf_64B        
-------------------------------------------------------------------------------------
```

And the following with `eyeriss = Eyeriss(Eyeriss_config(DIM=4,mem0_kb=8, mem1_kb=64, mem2_kb=1024))`:
```
********* Temporal Mapping - CostModelEvaluation(layer=LayerNode_0, core=1) *********
 O (O): [[], [('B', 5), ('K', 5), ('K', 5), ('C', 5), ('C', 5)], [('B', 20)], []]    
 W (I2): [[('B', 5), ('K', 5)], [('K', 5), ('C', 5), ('C', 5), ('B', 20)], []]       
 I (I1): [[('B', 5), ('K', 5), ('K', 5)], [('C', 5), ('C', 5), ('B', 20)], []]       
                                                                                     
-------------------------------------------------------------------------------------
 Temporal Loops                  O                  W                  I             
-------------------------------------------------------------------------------------
 for B in [0:20)                 sram_1M            sram_64KB          sram_1M       
-------------------------------------------------------------------------------------
  for C in [0:5)                 sram_8KB           sram_64KB          sram_1M       
-------------------------------------------------------------------------------------
   for C in [0:5)                sram_8KB           sram_64KB          sram_1M       
-------------------------------------------------------------------------------------
    for K in [0:5)               sram_8KB           sram_64KB          rf_64B        
-------------------------------------------------------------------------------------
     for K in [0:5)              sram_8KB           rf_64B             rf_64B        
-------------------------------------------------------------------------------------
      for B in [0:5)             sram_8KB           rf_64B             rf_64B        
-------------------------------------------------------------------------------------
```

This makes sense to me: smaller local buffers -> more data in the DRAM and more memory accesses, but it is not reflected in the number of cycles.

Also looking at the json representation (`cme[0][0].__jsonrepr__()`) for the generated cmes:
For the mapping I get the following with `eyeriss = Eyeriss(Eyeriss_config(DIM=4,mem0_kb=1, mem1_kb=2, mem2_kb=4))`:
```
{'outputs': {'memory': {'utilization': None,
   'word_accesses': {'O': [4waydatamoving (rd /\: 4000000.0, wr V: 4000000.0, rd V: 4000000.0, wr /\: 4000000.0),
     4waydatamoving (rd /\: 25000.0, wr V: 25000.0, rd V: 125000.0, wr /\: 125000.0),
     4waydatamoving (rd /\: 9000.0, wr V: 9000.0, rd V: 9000.0, wr /\: 9000.0),
     4waydatamoving (rd /\: 0, wr V: 0, rd V: 50000.0, wr /\: 50000.0)],
    'W': [4waydatamoving (rd /\: 0, wr V: 1600000.0, rd V: 4800000.0, wr /\: 0),
     4waydatamoving (rd /\: 0, wr V: 5000.0, rd V: 100000.0, wr /\: 0),
     4waydatamoving (rd /\: 0, wr V: 0, rd V: 10000.0, wr /\: 0)],
    'I': [4waydatamoving (rd /\: 0, wr V: 1600000.0, rd V: 4800000.0, wr /\: 0),
     4waydatamoving (rd /\: 0, wr V: 9000.0, rd V: 10000.0, wr /\: 0),
     4waydatamoving (rd /\: 0, wr V: 0, rd V: 50000.0, wr /\: 0)]}},
  'energy': {'energy_total': 215585000.0,
   'operational_energy': 1000000.0,
[... 56 lines cut ...]
    'W': [4waydatamoving (rd /\: 0.0, wr V: 2400000.0, rd V: 4800000.0, wr /\: 0.0),
     4waydatamoving (rd /\: 0, wr V: 125000.0, rd V: 2000000.0, wr /\: 0),
     4waydatamoving (rd /\: 0, wr V: 0, rd V: 10000000.0, wr /\: 0)],
    'I': [4waydatamoving (rd /\: 0.0, wr V: 480000.0, rd V: 4800000.0, wr /\: 0.0),
[... cut ...]
    'stalls': 0.25000300003600046,
    'stalls_onloading': 0.24508170043565722,
    'stalls_onloading_offloading': 0.2402515529860385}}},
 'inputs': {'accelerator': Accelerator(Eyeriss),
  'layer': [0, 1],
  'spatial_mapping': None,
  'temporal_mapping': None}}
```

I notice the 4waydatamoving is higher is the configuration with smaller buffers (again, this makes sense to me), but that is not reflected in the cycles, only in the power.

I am not sure what you mean by "fixing the temporal mapping"?

Best Regards,
Lukas

JiacongSun, 2024-05-26:
Hi Lukas,

Thanks for the more detailed information.

What I mean is you can adjust the temporal loop orders for both cases to understand the results, though I believe your current information is sufficient for this purpose.

It makes sense that the increased memory access count is not reflected in the total latency, as its latency can be hidden by the computation latency (through double-buffering). Therefore, the computation latency does not change in both cases, since the PE array size is the same.

In short, the latency difference between the two cases is due to the difference in on/off-loading latency. On/off-loading latency is calculated based on the required data amount of lower-level memories. In the second case (with bigger memory sizes), all loops are unrolled within SRAM, so the entire workloads/layers have to be loaded from DRAM to SRAM, which dominates the on/off-loading latency. However, in the first case (with smaller memory sizes), only partial workloads/layers are loaded from DRAM to SRAM during the data on/off-loading process. The way of processing data on/off-loading is hardware-dependent, so you may need to consider how to calculate the cost of this component if this is not your case. You can check how the on/off-loading latency is calculated in the code [here](https://github.com/KULeuven-MICAS/zigzag/blob/d7c174337dac61269f1bfdeb2682b3fb8f7c4c39/zigzag/cost_model/cost_model.py#L872).

Here is an intuitive explanation for your case. In your first setting (with smaller memory sizes), the amount of weight data required to be loaded from DRAM to SRAM_64KB is calculated as C_total×K_total/C(dram)/K(dram) =400. This means the required cycles for weight on-loading to SRAM is 400×precision/dram_portwidth=200. In the second setting (with bigger memory sizes), no for loop is unrolled on DRAM, resulting in the required weight data amount by SRAM becoming 10,000. This causes a latency cost of 10,000×precision/dram_portwidth=5000. The calculation in the code is more complicated, as it also considers factors such as whether a memory port is shared, multiple levels of memories, and the ceiling nature of latency calculation.

Best regards,
Jiacong

RobinGeens, 2024-08-06:
Hey all,
I have recently also encountered this phenomenon and Arne and I have identified this as incorrect behavior, inconsistent with the intent of the data loading latency. We will discuss a solution this week.

asyms, 2024-09-17:
This issue is fixed in https://github.com/KULeuven-MICAS/zigzag/pull/76.

The problem lies in the fact that loading and offloading phase of first/last input/output tiles would never overlap with the computation phase. In some scenarios where the inference is bandwidth bound and increasing the memory capacity doesn't improve data reuse, the cost model would thus predict a higher latency if now more data can fit in lower level memories.

We have fixed this by computing the amount of data loading/offloading that can be performed during the computation phase, taking into account the available bandwidth and required minimal amount of data to load/offload to actually start the first/last iteration.

closed 2024-09-17 by asyms