KULeuven-MICAS/zigzag #80: Example for user-defined temporal dataflow

chensy7, 2024-09-03:
Hello,

Thanks as always for open-sourcing and maintaining the great work. I'm wondering if there are any examples available for user to provide a fixed/constrained temporal mapping? Our use case would be that we want to provide some constraints on temporal loop orderings and allow Zigzag to tile/assign loop bounds as it sees fit based on the on-chip SRAM sizes, etc. Some accelerators do not have a lot of flexibility in case of temporal loop orders and we would like to evaluate those. 

Is this achievable on Zigzag now? I saw a few pull requests #75 #69 that seem to be relevant but I'm not sure about the correct usage. Any help/guidance on this would be much appreciated.

Thank you!

RobinGeens, 2024-09-12:
Hello!

Your description sounds exactly like the feature that has been added in https://github.com/KULeuven-MICAS/zigzag/pull/75, but the examples and tests that use the constrained mapping seem to be missing from the master branch. I will get back to you with an example input ASAP!

asyms, 2024-09-13:
Also, you can refer to https://github.com/KULeuven-MICAS/zigzag/pull/82, where we're working on integrating the tests using constrained temporal mapping.

chensy7, 2024-12-14:
A gentle ping on this thread. I tried the constrained mapping yaml in #82, but they are not working for me. Would appreciate any help!