KULeuven-MICAS/zigzag #85: lab 4 in tutorial-new branch is giving errors

Mehrozkhan, 2024-09-12:
I am trying to use zigzag for IMC case and using the labs to learn it. I could find the videos for lab 1 to lab 3 but I could not find any video for later labs. 
Also, when I try to run the main file in lab 4, it throws an error. I am pretty sure I am doing something wrong here.
Here is the snippet from error
``Traceback (most recent call last):
  File "/home/muhamadmehrozkhan/zigzag/./lab4/main.py", line 15, in <module>
    accelerator = AcceleratorParserStage.parse_accelerator(hardware)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/muhamadmehrozkhan/zigzag/zigzag/stages/AcceleratorParserStage.py", line 30, in parse_accelerator
    accelerator_data = open_yaml(self.accelerator_yaml_path)
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'accelerator_yaml_path'``

JiacongSun, 2024-10-01:
Hi @Mehrozkhan ,

The IMC lab is not merged within this branch, so it does not work there.

If you want to run lab4-6, you can open an older version of zigzag (commit d7c1743 on May 22th, 2024) at: https://github.com/KULeuven-MICAS/zigzag/tree/d7c174337dac61269f1bfdeb2682b3fb8f7c4c39. Lab4-6 should work for you then.

For lab7-9, please keep using the `tutorial-new` branch.

Let us know if your problem is not solved!

Best regards,
Jiacong

Mehrozkhan, 2024-10-01:
Hi @JiacongSun , Thanks.  I figured that out already.

closed 2024-10-01 by Mehrozkhan