KULeuven-MICAS/zigzag #98: About hardware mapping files.

0x7D6E7FD, 2024-11-11:
Hello ,

I just explored this repo a few days ago. Here ,[https://github.com/KULeuven-MICAS/zigzag-imc)]( https://github.com/KULeuven-MICAS/zigzag-imc) in this link , I saw some of the DIMC has been evaluated in this framework. can you kindly point me to the hardware files and mapping files for these DIMC papers ?
just like I am seeing files inside `zigzag/inputs/examples/hardware` and` zigzag/inputs/examples/mapping/tesla_npu_like.py ` in these folders. I want to see the these files for DIMC mentioned in this figures.
[image]

asyms, 2024-11-15:
Hi, thanks for your interest!

@JiacongSun could you have a look and point her to the correct settings please

JiacongSun, 2024-11-15:
Hi,

These chips are used for validation through specific scripts. Please have a look at https://github.com/KULeuven-MICAS/zigzag-imc/tree/master/imc_model_validation for the details.

To your question, they are not provided as hardware templates, so there are no valid templates yet. However, using the parameters in the table, you can fill these numbers into the IMC template files of [DIMC](https://github.com/KULeuven-MICAS/zigzag/blob/master/zigzag/inputs/hardware/dimc.yaml) and [AIMC](https://github.com/KULeuven-MICAS/zigzag/blob/master/zigzag/inputs/hardware/aimc.yaml).

asyms, 2024-11-15:
Thanks @JiacongSun!

Closing this issue now. Feel free to open any other issues if you have more.

closed 2024-11-15 by asyms