KULeuven-MICAS/zigzag #99: Possible issues in get_act_layer_op and get_weight_layer_op

yuli821, 2024-12-12:
Hi, I'm using Zigzag to analyze a customized vision transformer model and trying to map it onto Digital IMC architecture. One thing I observed is that when there are no constant operands, get_act_layer_op always selects the second input operand (layer_node.py, line 206) and get_weight_layer_op always selects the first input operand. I printed out the selected operand and found out the returned act_layer_op is W and weight_layer_op is I. This raised an issue in calculate_mapped_rows_when_diagonal_mapping_found since the act_pr_layer_dims will always be empty if act_layer_op is W. 

I simply switch the index to resolve the error, but I'm not sure if the logic is correct here. Why does the original code select the second input operand for act_layer_op and the first operand for weight_layer_op? Is there a specific reason behind it? I'm not sure if this fix is valid here.

I have another problem in evaluating the TOP/s of the IMC macro and the overall TOP/s. In the function, get_macro_level_peak_performance, tops_peak is given by mac*2/clock_cycle_period/1000 while the clock_cycle_period is in ns. Should it be mac*2/clock_cycle_period/1000000000 if we want to compute TOP/s? I'm also confused on the way that lab6 (main.py) computes the overall latency, delay_in_ns = energy * cmes[0].tclk. I printed out cmes[0].tclk, which looks like the overall clock cycles. Why does the code multiply energy with the total clock cycles to get the delay? Should it multiply the clock cycle with the clock period?

asyms, 2024-12-23:
Hi, thanks for bringing this issue to our attention!

I think that the case where there are no constant operands might have been an oversight, as we traditionally have focused mostly on CNN workloads where you always have weights.

@JiacongSun do you have any insights regarding this?

JiacongSun, 2024-12-23:
Hi yuli,

Thanks for bringing up this issue!

This issue is similar to the one reported in https://github.com/KULeuven-MICAS/zigzag/issues/90, which was identified beforehand. Unfortunately, we have not had the manpower to fix it.

Your current method is correct and should work for your experiments. You can try revising the code to cover more general cases. If possible, we welcome you to make a pull request after fixing it.

Regarding lab 6, it is under the draft version and full of errors, such as the one you noticed on the delay calculation. Sorry for the confusion. We will update and publish it later with a README document. You're welcome to review it after that, but please disregard it for now.

If you do need a lab for IMC, please have a check on lab5. It is already cleaned and should serve your purpose.

JiacongSun, 2024-12-30:
The lab6 has been updated now. Please have a check.